emmebrusa/TSDZ2-Smart-EBike-860C #7: The used Wh for the trips are not saved in EEPROM

22jj, 2022-06-23:
Instead an offset is saved that only is set when a manual SOC reset is executed. The used Wh is always 0 at startup.
A fix would be the replace the lines in firmware/common/src/eeprom.c

```
m_eeprom_data.ui32_wh_x10_trip_a_offset = ui_vars->ui32_wh_x10_trip_a_offset;
m_eeprom_data.ui32_wh_x10_trip_b_offset = ui_vars->ui32_wh_x10_trip_b_offset;
```

with

```
m_eeprom_data.ui32_wh_x10_trip_a_offset = ui_vars->ui32_wh_x10_trip_a;
m_eeprom_data.ui32_wh_x10_trip_b_offset = ui_vars->ui32_wh_x10_trip_b;
```

I also added an option to reset the trips on recharge in my [fork](https://github.com/22jj/Color_LCD_860C) if you want to take a look.
It only works in the Wh SOC mode not the auto or volt mode.

emmebrusa, 2022-06-24:
Trip reset at full charge is already provided, by setting trip reset hours to zero. It's not the same?
As for the Wh used in trips, in what conditions and with what settings does the problem occur?
I have never noticed it, however I will check. Thank you.

22jj, 2022-06-24:
It seems like the functionality for trip reset should be the same. I think it is just unintuitive. Also I noticed you just forgot to handle the trip reset if a manual soc reset occurs (in BatterySOCReset() in mainscreen.c).

The problem always occurs. ui32_wh_x10_trip_a_offset is only ever written to in the BatterySOCReset function.

emmebrusa, 2022-06-24:
I did a quick check, the lines you edited they are in the file firmware \ 860C_850C \ src \ lcd.c lines 98 and 99, they are in that position because they must be saved only at power off, putting them in eeprom.c would also be saved when exiting the configuration menu.
If you've noticed a problem, I think it's another cause.
Unfortunately in this period I don't have the bike with 860C available, I can't do tests.

22jj, 2022-06-24:
It is not present in the version you uploaded: https://github.com/emmebrusa/Color_LCD_860C/blob/master/firmware/860C_850C/src/lcd.c
Maybe you didn't push the current version.
I tested it with putting it in eeprom.c and it works without problem because the EEPROM is only read at boot.

emmebrusa, 2022-06-25:
Azzz ... you are right, in the uploaded version there is not, something went wrong in the push.
Soon I will be doing an update for other small bugs, I will add this too.
The missing lines in lcd.c:

`void lcd_power_off(uint8_t updateDistanceOdo)
{
  // save current battery Wh
  ui_vars.ui32_wh_x10_offset = ui_vars.ui32_wh_x10;
  
  // save current trip A and B Wh
  ui_vars.ui32_wh_x10_trip_a_offset = ui_vars.ui32_wh_x10_trip_a;
  ui_vars.ui32_wh_x10_trip_b_offset = ui_vars.ui32_wh_x10_trip_b;`

Now everything is clear, no one has ever noticed the problem because the bin files are correct.
You noticed it, because you tried to fill in, thanks for the report.

In eeprom.c it only works fine if you don't make any changes in the configuration menu.