emmebrusa/TSDZ2-Smart-EBike-860C #8: Add support for executing mspider65 hall sensor calibration

Ornias1993, 2022-10-18:
The Hall Sensor calibration code by mspider65 is mostly already part of the firmware.
All we need at this stage, is a way to actually execute it, for which mspider65 uses the ESP32.

Worst case scenatio, we can at least give configuration fields to manually enter the configuration parameters and users can use the esp32+firmware by mspider65 to fetch their values.

emmebrusa, 2022-10-23:
Entering these parameters manually is not a good idea, a mistake could cause serious damage to the controller.
For hall sensor calibration, the necessary code is not only in ESP32, there are very complex calculations also in the android app.
Everything should be done with the display, I think it's possible but I don't know if I'll do it.

iamblacklite, 2022-11-20:
I have implemented this in a very haphazard way - it's time consuming and not at all intuitive, and I'm not sure that the results are actually worth it. And of course the potential for damage is high. Minimising and maximising values needs to be done by hand with the bike on a stand, and then lots of values recorded and then entered into a text based app that uses the algorithms from mspider65's Android app. These values need to be entered back into the display. Then the motor current needs to be minimised at different erpms by adjusting the rotor angle offset in real-time, and values plugged into a spreadsheet, some mathematics done on the values etc.. It was an interesting experiment but without the capabilities of an Android app to automate the functions I think it's far beyond what the average user would be interested in doing.