emmebrusa/TSDZ2-Smart-EBike-860C #10: Display only stays on when on button is pressed

elpollodiablo, 2023-03-18:
Hello,

after flashing and using this firmware, I experienced the following problem:

- The display would not power on normally
- The display would only stay powered while pressing the "on" button, but immediately switch off when i let the on button go
- I could reflash while powering the display via the on button, but no firmware would help

I wrote an email to the developer of the display, aptdevelop.com, if they could help me identify this problem, and I received the following answer:

"According to your description, component(Q2) was damaged, you can refer to the following pictures. Q2 (2SA1201) is a power supply switch (triode), it supply power to controller and headlight module. The Q2 was burned, it could only be caused by overcurrent, wrong wiring sequence, controller or headlight module failure could lead to this result."

This led me to opening the display with a heating gun loosening the glue around the display, and then inserting a thin plastic spatula in the rim, basically the standard procedure opening glued mobile displays (such as the iphone). There is no wiring around the rim of the display, so it's relatively safe.

Lo and behold, Q2 actually was damaged (it seemed to have exploded in my case). The 2SA1201 part is in SOT-89 format, I ordered the "DY" version on ebay and will report back if it worked after I received and soldered the replacement.

This is not a bug report, but an attempt to document this in a place where I looked for an answer :

Hope that helps people having a similar problem in the future.

Thanks for this awesome firmware everyone working on this btw!

stefesciberras, 2024-04-29:
Have exactly the same issue... so it is a display issue?

elpollodiablo, 2024-04-29:
> Have exactly the same issue... so it is a display issue?

If the symptoms are exactly as I described, the probability is very high that it's the same issue - you probably accidentally connected 36v/48v to the 5v rail. Unfortunately, the only way to confirm is to open the display and verify the damage visually or electrically.

Here is the image the helpful person at aptdevelop.com sent me: https://i.imgur.com/HkRzMkO.png - in my case, half of the transistor was missing instead of melted :)

stefesciberras, 2024-04-29:
I have opened the display up, and to be honest, there seems to be nothing actually wrong physically. I am now worried: it is either a defective unit, or there is a problem with the hardware. 
I cannot seem to find the older versions I had previously flashed.

SykloFR, 2024-04-29:
I think you have the new PCB version V1.3, wich comes with a 2 positions
Power button, so when press and released it send 2 time Power function so
On then Off.
Try flashing V1.3 instead.

stefesciberras, 2024-04-29:
I will try the following:
[860C_V13-v20.1C.5-beta-1-bootloader.bin](https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/blob/master/releases/v20.1C.5-beta-1-860C/860C_V13-v20.1C.5-beta-1-bootloader.bin)

is this correct? I am assuming the V13 refers to the PCB V1.3?

thanks again