emmebrusa/TSDZ2-Smart-EBike-860C #11: Trip Data, Odometer not saved if battery dies!

Chonny87, 2023-05-14:
If your battery dies it doesn't save/log your last trip. It appears the odometer and trip data only save at shutdown. Therefore a loss of battery power without properly shutting down your odometer, setting changed and trip data don't get saved! The controller should save trip data, odometer and setting (if changed) when on intermittently. Example every 5 minutes so your trip data is saved if your battery dies or your battery gets unplugged without shutting down.

Ploon, 2023-05-14:
I would rather it stayed as it is. The non volatile memory within the display has a finite number of writes before failure and constantly updating for the unlikely event that a user fully drains the battery ( or forgets to shut down properly before switching off battery), would result in premature (and costly) display failure and replacement. It is usual and sensible engineering practice to save such data on shutdown.

Chonny87, 2023-05-14:
You answered my question, I did not know this. Thank your for sharing this! This is closes as far as I am concerned

closed 2023-05-14 by Chonny87

Ploon, 2023-05-14:
You are very welcome, As you didn’t know about the memory life, I completely understand why you brought it up 😊👍