emmebrusa/TSDZ2-Smart-EBike-860C #14: Clock is very inaccurate

cortado, 2023-08-08:
It only takes a month for the clock to be many minutes off. Firmware needs update, or other fix?

emmebrusa, 2023-08-10:
My 860C goes forward 5 minutes/year.
Acceptable tolerance.
I have never delved into the management of the watch, I don't know if a calibration is foreseen.
If you are able to compile, you can try changing the value of:
#define SECONDS_IN_DAY 86399
in the rtc.c file.

jim-sharratt, 2023-08-12:
FYI - My 860C goes forward as well, last updated at the end of March (when the clocks changed) and today it's 100 minutes forward.

cortado, 2023-08-12:
Ok, sounds like a correct clock is something I can't expect. No big problem, but just found it a little strange, but maybe more complicated than I think.