emmebrusa/TSDZ2-Smart-EBike-860C #16: Tension increase with power consuption

Oregos29, 2023-09-02:
Hi,
Thanks for your OSF.

I notice that the display show a tension which increase proportionally with the power consumption of the motor.
For example, going without pedaling : 48v
If the motor is about 300w : +1.V : 49v
If the motor is about 600w : +2v = 50v
Full power 730w : +2.5v = 50.5V.
I stop pedaling, the tension decrease to return to 48v.

With the original firmware, this was inverted :  The tension decrease with power consumption.
Configuration : TSDZ2B 750W 48V
Regards

emmebrusa, 2023-09-21:
Voltage fluctuation compensation is excessive.
Try decreasing the "Resistance" value in the "Battery" menu.

Oregos29, 2023-09-21:
Thanks. Sorry to post here for a parameter mistake.