emmebrusa/TSDZ2-Smart-EBike-860C #17: Suggestion : Slave light control to light sensor of the display

Oregos29, 2023-09-02:
Hi,
Thanks for your OSF.

I have a suggestion (I hope to be in the good place for that) : 
Have the possibility to slave the light control of the 860C to its light sensor.
So, the light could be automatic power on under a level of light preset.
This fonction is on the original firmware of the 850C.

Regards,
GREG

emmebrusa, 2023-09-21:
Thanks for the tip, I'll watch this winter when development resumes.

Oregos29, 2024-01-22:
Hi, thanks to have donne it.
Greg

Oregos29, 2024-11-10:
Hi,
Auto on/off lights with the display's light sensor (860c only) works well, thanks. But when luminosity is about limite value to Switch on/off, the light is flashing. May be have two limits values will solve the problem : a value to Switch OFF when the light is ON,  another value to Switch ON when the light is OFF. ( Or a single Switch value and a trigger)
Brest Regards

emmebrusa, 2024-11-12:
Does the light flash even when setting "Light Hysteresis %" to 20?

Oregos29, 2024-12-01:
Hi , in the version 20.1.C.4, i dont have "Light Hysterisis %" parameter. I juste have "Light sensitivity %".
I dont see anything about automatic light in the comments of last releases. Do you add this parameter since the .4 release ?
Regards

emmebrusa, 2024-12-02:
"Light Hysterisis %" was added with version 20.1.C.5, unfortunately the manual is not updated yet.
[https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases/tag/v20.1C.5-860C](url)
I hope to do it with the next version.
There will also be an improvement to the automatic on/off of the lights, I added a delay to the lights turning off.

Oregos29, 2024-12-04:
I try to flash my firmware but impossible for the motor : In the soft STVP, i have the message "file : line 7 : No in inel Hex format" when i try to load the .hex file. Older .hex files works.

emmebrusa, 2024-12-05:
Maybe the download failed?
Or did you download in ASCII format...
The file TSDZ2-v20.1C.5-860C.hex is 48.7 KB?

Oregos29, 2024-12-05:
File was about 250kb... bad download (save target). Works fine now, flashing ok. I let you know for Hysterisis. Thanks for all.

Oregos29, 2024-12-10:
Hysterisis works well, thanks