emmebrusa/TSDZ2-Smart-EBike-860C #19: Power assist level calcluations?

bogardon, 2024-06-27:
> “Power assist” is an assistance mode proportional to
the power on the pedals.
Levels available from 1 to 9. Set assistance levels
according to your needs. Value% / 2, maximum 254.
For example, applying 100 Watt to the pedals, with
>150 assist level, the motor delivers 300 Watt.
These assistance parameters are also used in the
hybrid mode.

Can somebody explain this calculation to me? I don't understand how to derive 300W from 150 assist level and 100W human power. Thank you.

emmebrusa, 2024-06-28:
Assistance levels can be set from 1 to 254.
This value is half the percentage of multiplication of human power.
So if set to 150, 150 x 2 = 300%.
With human power 100W, motor power 300W.

closed 2024-06-28 by bogardon

hydriniumh2, 2024-11-24:
I think that means it should be `value % * 2` not /, though?