emmebrusa/TSDZ2-Smart-EBike-860C #22: fw v20.1C.5-860C-update-1 LM35

freexxxx, 2025-03-09:
Hi a update to latest FW v20.1C.5-860C-update-1 but i have start form cold tsdz2 with lm35 temp is (start) from 33°C temp. I think it is no good temperatre. On old FW TSDZ2-v20.1C.4-2-860C was temp OK. It is bug or?

emmebrusa, 2025-03-09:
Are the parameters in the "Motor temperature" menu set correctly?

[image]

freexxxx, 2025-03-15:
yes I have

[image]

[image]

emmebrusa, 2025-03-16:
The value 33/34 is displayed when no sensor is connected, or if "Motor temperatire->Feature" is set to "disable".
I saw that you set Units=celsius, have you tried auto?
Since version 4.2, I have only added support for the temperature sensor type TMP36.
So it is unlikely to be a bug, more likely a hardware problem.
A clarification, I make sure that the custom settings are maintained when updating from the previous version, they should be maintained even with older versions, but I cannot guarantee it 100%.
So if you have not already done so, try, in the "Display" menu, to do a "Reset to default" and "Confirm reset".
Be careful, your custom parameters will be lost, take note of them first, especially Odometer and Charge cycles.

Regarding the maximum speed, the motor power has not changed, there must be another explanation, check the maximum power in the "Bike" and "Motor" menus.

freexxxx, 2025-03-17:
I have 860C V5.2 it is older I use 860C_v20.1C.5-1-860C-bootloader. I try what you wrote a no change. If I set Motor temperatire->Feature" is set to "disable". the temperature is 0. I flash back to v20.1C.4-2 and temp is OK.

emmebrusa, 2025-03-17:
If it doesn't work with 5-1 and reflashing 4-2 works, it's clear that there's something wrong with the firmware.
First of all I correct myself, the value 33/34 is displayed when no sensor is connected and with "Motor temperature->Feature" set to "temperature".
Unfortunately I can't try 5-1 on the bike where I installed LM35, it's been stopped since November due to a broken connector, I don't use it in the winter.
However, I used version 5-0 all last summer until November and the temperature worked correctly.
Today I tried on the other bike where there is no LM35, simulating the ADC input by setting the value on the display, it works.
So I reserve the right to try, for confirmation, on the bike where there is LM35 when I have fixed the connector, I will keep you updated.