emmebrusa/TSDZ2-Smart-EBike-860C #28: Info about sw102

lorenzvee, 2025-07-03:
Hi,

I hope you can give me a tip to resolve my issue with my tsdz8 setup because mstrens recommended me to ask you.

I can correctly use the 860c display without problem but when I try a sw102 flashed with the osf firmware ( tried all the fw on github), I always get a 6 fatal error after the message of the assist mode init.

I tried various config to reflect the 860c without result.

Is there a way to find the exact cause of this error? (By edit the source and recompile?)

You can find a photo of the error, in this case the battery was not correctly calibrated in the menu.

[image]

Thanks

mstrens, 2025-07-18:
Is version TSDZ2-v20.1C.5-update-1-860C valid for a SW102?
It contains 860C in the file name. So, I expect it is for 860C only.
Still this page https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases/tag/v20.1C.5-update-1-860C says that it has not been tested for SW102. 
It is not clear if it can be used for SW102.
If not, does it exist a similar version for SW102 and where to find it? 
If it exists, has it been tested for TSDZ2 ?

emmebrusa, 2025-07-18:
The file for SW102 is sw102-otaupdate-20.1C.5-1.zip
You can find it in the latest version
[https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases/tag/v20.1C.5-update-1-860C](https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases/tag/v20.1C.5-update-1-860C)
Which is identical to sw102-otaupdate-20.1C.5-1-860C.zip found in the repository; I just forgot to rename it.
Keep in mind, however, that I don't have a SW102 and have never tried it; I simply modify the code shared with the 860C and compile. From the feedback from those who use it, it works with the TSDZ2.
Regarding the "e 6 Fatal error," it's probably a communication timeout issue, but again, I've never looked at the SW102-specific code and I don't have a display to test.

emmebrusa, 2025-08-02:
The sw102-otaupdate-20.1C.5.zip file in the Color_LCD_860C repository is identical to the one found in the release [https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases/tag/v20.1C.5-860C](https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases/tag/v20.1C.5-860C).

If you still get the e6 Fatal error with Undervoltage disabled, I'm not sure what to think.
The code that generates the error is identical in the latest version; I haven't changed anything.

emmebrusa, 2025-08-02:
It could be that, having increased the lines of code, the initialization time is insufficient.
Perhaps just increasing the time from 20 to 30.

```
	if (ui8_comm_error_counter > 20) {
		motor_disable_pwm();
		ui8_motor_enabled = 0;
		ui8_m_system_state |= ERROR_FATAL; // Comms failed
	}
```

Unfortunately, since I don't have a display or a motor, I can only speculate.

Badpioneer, 2025-08-02:
@mstrens
The latest fix successfully resolved the issue!
Increasing the initialization time eliminates the error.

mstrens, 2025-08-02:
Fine.
I put it on gitrhub as version 00.01.32