emmebrusa/TSDZ2-Smart-EBike-860C #35: Setting the min and max values for ADC throttle

mstrens, 2025-08-21:
In VLCD5 version, it is possible to edit the values used to remap the ADC throttle.

It seems that there is no such fields in 860C (and SW102) version. There are min and max values in temperature menu but it seems there are only used  for temperature limits in TSDZ2 OSF code.

Still some throttle (e.g. Bafang one used with 860) have other  range than the one defined in #define.

It could be useful to allow to edit those fields on 860C (and SW102) display.

Still it does not require absolutely to add more fields on the screen.

As the ADC is used or for temperature or for throttle, the min and max value fields could be used for both.

This avoid also to add new fields to the frame transmitted to the controller.

Doing so, it would be enough to change only some code in the controller (and not in the 860C firmware). Still it would be usefull to adapt the user manual to explain that fields could be used for temperature OR throttle.

Please let me know if you plan : 
- to let 860C firmware unchange : In this case, I think that, in controller version for TSDZ8, I will use the min and max temperature limits to remap ADC throttle too
- to use the existing min and max fields for temperature  and throttle.  In this case,  in controller version for TSDZ8, I will use the min and max temperature limits to remap ADC throttle too.
- to add 2 new fields in 860C (SW102) display firmware and to the frame sent to the controller. Then I will use those new fields in TSDZ8.

emmebrusa, 2025-08-21:
Sure, you can use the same min and max temperature fields; they're alternative functions.
Go ahead.

emmebrusa, 2025-08-28:
Ok to the next release.